EnhancedRadioDevices/915MHzEdisonExplorer #1: Suggestion: UART port

sulkaharo, 2016-08-28:
If I understood correctly, the board uses the SPI to talk to the Edison. Having the second UART port exposed on the board would be huge for adding further features, such as connecting a Wixel for 2.4GHz band capture. Would this be in the scope for this board, or would you suggest for example sandwiching this with the Sparkfun UART board?

mogar, 2016-08-28:
Do you mean from the Edison to the CC1110? Yes, that's a SPI port. There are not really any GPIO broken out from the Edison that aren't already used on the board, so if you want to use a UART with an external board like the Wixel I recommend using another Edison Block.

That said, the Edison's UART and OTG are exposed through USB connectors. It would probably be possible to connect the OTG port to something like the wixel. I would also think that it should be possible to use the Edison's own 2.4GHz radio to do a capture in that band, but I haven't looked into it.

mogar, 2016-09-15:
We could potentially break out some GPIO or another UART to a header for this purpose. I'll look into it for the next revision.

oskarpearson, 2016-09-15:
@sulkaharo Are the devices we're thinking of using in this way all 3.3v?

If so, would we also need to break out a ground and 3.3v points to supply power to the UART device?

Also, would the devices require flow control (RTS/CTS)?

If so it'd mean breaking out 6 pins: TX, RX, RTS, CTS, +3.3v, GND - is that correct? Would these require through-hole soldering connectors?

mogar, 2016-09-15:
The Edison has two UARTS. UART2 is used for the USB-UART converter. UART1 could be used for this, and it does have flow control.

We could use a tiny connector for these so that people wouldn't have to solder them. I don't expect we'd have a ton of space, so standard 0.1" headers won't work.

Given how tight the routing is getting on the Explorer board, it might be worth having a second mezzanine board between the Edison and the Explorer that actually breaks out the pins. That would make the whole thing about 5mm thicker, but would be easier to connect ti.

sulkaharo, 2016-09-15:
I'm thinking of the Wixel, which should work fine with 3.3. Exposing 1.8v level would allow hooking a logic level shifter in the mix?

mogar, 2016-09-15:
The Edison pins are 1.8V by default, so we'd need to either throw another level shifter on the board or use one off-board to communicate with the Wixel. At this point, I'm thinking that this is going to effect board size too much, and it might make sense to just use a second Edison Block to access those pins.

mogar, 2016-10-11:
There's not really space on the board for another level shifter. I'm going to say that we won't be adding a second UART, and if you want one I recommend using one of SparkFun's adapter blocks between the Explorer and the Edison.

closed 2016-10-11 by mogar