EnhancedRadioDevices/915MHzEdisonExplorer #2: Plugging/unplugging power power-cycles the Edison

scottleibrand, 2016-08-28:
When the power switch is ON, plugging in microUSB power (to either the UART or OTG microUSB port) power-cycles the Edison.

It would be *much* preferred if we could have the Edison stay powered while transitioning from battery power to charging.  I would even give up the hardware power switch entirely for that if necessary.

mogar, 2016-08-28:
This looks like it won't be easy to fix on the current board rev, but it should be easy to fix for the next hardware rev.

mogar, 2016-08-28:
With the current hardware rev, it's possible to cut a trace that will cause the board to not accept power from USB. USB can still charge the battery, but it can't turn on the Edison or CC1110. This fixes the power-cycle issue at the expense of not being able to use it without a battery.

If you get me your hardware, I can make this modification for you. I'll also post some notes about how to do it next week.

scottleibrand, 2016-08-29:
Ok, thanks.  Definitely something we should fix in future versions.  I'll probably wait until we figure out antenna tuning before modifying this board.

mogar, 2016-09-15:
For the next hardware rev, we're going to switch the inputs (battery and USB power) to U9. We're also going to replace the 4V LDO that generates USB power to the Edison with something that can handle more current (probably the AP7365).

scottleibrand, 2016-09-28:
I've now got both Explorer boards (one with the trace cut, one without) fully up and looping well on battery power, but have been seeing them powercycle almost constantly (within minutes of boot) when plugged into USB.  Was the trace cut supposed to fix that, or is that an unavoidable issue until we get the 2nd gen hardware with the AP7365 or similar?

Will it be practical to upgrade the 4V LDO on existing boards, or would it be better to just use them as dev boards and replace them with the newer ones for production use?

mogar, 2016-09-28:
The cut trace should have fixed that issue. I'd like to take a look at your
hardware and see what's going on, since I've had on problems with the cut
trace on four different boards I've done it to.

On the topic of the LDO upgrade: I'd like to upgrade your hardware and have
you test it out. Long term, I recommend getting the newer hardware after
the fix is integrated for the stability. The upgrade is not going to be as
robust as a full standard circuit board.

mogar, 2016-10-11:
The design has been adapted and the next revision won't have this bug.

closed 2016-10-11 by mogar