EnhancedRadioDevices/915MHzEdisonExplorer #3: USB OTG

scottleibrand, 2016-09-15:
Has USB OTG been tested?  I can't seem to get our board to recognize a Dexcom (or a Kindle) connected over USB OTG.  `lsusb` does detect the OTG cable itself, though, as a USB 2 and a USB 3 hub.  Wondering if there may be some power draw limitations like the Intel breakout board has?

mogar, 2016-09-15:
USB OTG hadn't been tested yet.

There is no 5v regulator on the board, so if you connect a device that
needs external power then it won't work. Have you tried a powered hub?

mogar, 2016-09-15:
If we reverse D5, then the USB OTG port can be powered from the UART port. This would allow you to plug the OTG device directly into the board without using a powered hub.

scottleibrand, 2016-09-15:
So that would require external USB power then?  Do I correctly assume that powering the USB OTG port from a JST-connected lipo would be infeasible?  Or is that an option too?

Pass-through USB power from UART to OTG port would probably be OK: we'd just need to plug in an external USB battery for offline mode, but if we're plugging in a USB device we're already giving up on the tiny form factor, so that's not as big a deal.

mogar, 2016-09-15:
Powering the USB OTG port from a connected LiPo is possible, but requires additional circuitry to boost the voltage to ~5.2V. That will add size, cost, and potentially thermal issues. Could be worth it though.

I think using an external battery for passthrough from the UART to the OTG port is probably the best way to go.

PieterGit, 2016-09-30:
Does a Dexcom G4 require a USB OTG connection to be connected to the Explorer Board or is it possible to use a Dexcom G4 with a USB connection with the Explorer Board?

Being able to use wire up a Dexcom G4 (instead of using blueotooth) would be a really nice feature.

scottleibrand, 2016-09-30:
I believe the next round of hardware will support connecting the Dexcom G4
over USB OTG, but only if you also plug in external USB power to the UART
microUSB port.

One of the next things on my list is to see if I can get openxshareble
working with oref0-setup on the Explorer board with jubilinux...

Does that answer your question?  I'm not quite clear what you mean by "require
a USB OTG connection" vs. "use a Dexcom G4 with a USB connection".

PieterGit, 2016-09-30:
I mean: can I use the Dexcom G4 wired by USB/USB OTG with the Explorer board. Most bluetooth connection tend to be really pain in the ass to setup, and bluetooth libraries on Linux seem to have several quirks. 

So I would prefer to use a wired USB connection between the Dexcom G4 receiver and the Explorer board. I thought this was a "regular" USB-connection and thought USB OTG was a different thing.

I think the form factor of Dexcom receiver and a explorer board and lipo battery is still fairly acceptable, so I would prefer a way to power a rig like that by using only one battery somehow. Will that be possible?

scottleibrand, 2016-09-30:
>
> I mean: can I use the Dexcom G4 wired by USB/USB OTG with the Explorer
> board. Most bluetooth connection tend to be really pain in the ass to
> setup, and bluetooth libraries on Linux seem to have several quirks.
>

Yeah.  I'll be trying to figure out that setup and iron out the BT quirks,
but we'll see.

> So I would prefer to use a wired USB connection between the Dexcom G4
> receiver and the Explorer board. I thought this was a "regular"
> USB-connection and thought USB OTG was a different thing.
>

USB OTG is simply the way you do "regular USB" over a microUSB port.

> I think the form factor of Dexcom receiver and a explorer board and lipo
> battery is still fairly acceptable, so I would prefer a way to power a rig
> like that by using only one battery somehow. Will that be possible?
>

Unfortunately it's not possible to power the USB OTG port from a lipo
without a 5v upshifter, which would require significant board redesign and
expansion.  You will, however, be able to remove the lipo if you're
powering the Explorer board over USB. Or you can have a smaller USB battery
and keep the lipo.

PieterGit, 2016-09-30:
Thanks for clarifying USB OTG. I assume a 5v upshifter is similar to a AdaFruit PowerBooster module. 
Would it be possible to enhance the design of the board so that a user can add a upshifter/PowerBooster module as an extension so that the rig can be powered by one battery. 

Or am I right that a LIPO connected to a AdaFruit PowerBooster 1000 module, that powers the Explorer board over USB makes it possible to connect the Dexcom G4 with USB OTG (perhaps also connecting the 5V to the UART/USB OTG port).

This would probably also improve battery life a bit (using wired connection instead of Bluetooth). But main reasons for me for this use case will be easier setup and improved reliability. If you guys can think of ways to integrate this, or make it easier to integrate this as an add-on, I would be very very happy.

scottleibrand, 2016-09-30:
>
> Thanks for clarifying USB OTG. I assume a 5v upshifter is similar to a
> AdaFruit PowerBooster module.
>

Yes, or a similar chip onboard.

> Would it be possible to enhance the design of the board so that a user can
> add a upshifter/PowerBooster module as an extension so that the rig can be
> powered by one battery.
>

Yes, but we decided not to do that due to size constraints and potential
extra power drain.  If you want a larger board with that built-in (as well
as built-in wiring for wixel), the RiledUp board might be a good option.

> Or am I right that a LIPO connected to a AdaFruit PowerBooster 1000
> module, that powers the Explorer board over USB makes it possible to
> connect the Dexcom G4 with USB OTG (perhaps also connecting the 5V to the
> UART/USB OTG port).
>

Perhaps, although I'm not sure why you would bother with lipo + powerboost
rather than just getting a 5V USB battery.

> This would probably also improve battery life a bit (using wired
> connection instead of Bluetooth).
>

Unlikely: the USB with 5V upconversion drains significantly more power than
Bluetooth does, even if you turn of the Dexcom's charging feature.

> But main reasons for me for this use case will be easier setup and
> improved reliability. If you guys can think of ways to integrate this, or
> make it easier to integrate this as an add-on, I would be very very happy.
>

Yeah, that will likely be a common use case if we can't get openxshareble
reliable enough.  I'm working on that right now, so I'll let you know how
it goes.

PieterGit, 2016-10-04:
I believe a lot of (non US) Dexcom G4 users don't have the possibility to use the Share / bluetooth function. I think the hardware supports bluetooth, but the receiver firmware disables the share/bluetooth afaict. A way to wire up the Dexcom to the Explorer board somehow would really be an essential feature for those users.

scottleibrand, 2016-10-04:
I'll be meeting up with Morgan tomorrow to test using a Dexcom over microUSB OTG (with external 5V USB power on the other microUSB port).

scottleibrand, 2016-10-06:
We tested that the new Edison Explorer board can read BG from a Dexcom connected over USB, if it also has a USB battery providing 5V power on the other USB port. Not the most compact setup to carry an external battery, but allows for completely offline looping without reliance on openxshareble.

closed 2016-10-06 by scottleibrand