EnhancedRadioDevices/915MHzEdisonExplorer #7: Break out points for 3.3v / GND

oskarpearson, 2016-09-15:
It may be useful to break out the 3.3v and ground outputs for powering other devices. These could through-hole connectors, or simple surface-pads that are suitable for soldering.

I'm separating this from https://github.com/EnhancedRadioDevices/915MHzEdisonExplorer/issues/1 since it may be a separate issue.

mogar, 2016-10-11:
The space constraints on the board make this a bit tough to do in a way that beginning solderers can handle. The next revision won't have the SMT header populated for the CC debugger, and there's a 3.3V and GND pad on that footprint. You could use those if you really needed this. Think that'll be sufficient?

oskarpearson, 2016-10-11:
That's a-ok with me. Thanks!

closed 2016-10-11 by oskarpearson