EnhancedRadioDevices/915MHzEdisonExplorer #10: Power and OTG

glenrob1940, 2016-12-26:
Using the board with openAPS.  Attempting to go offline using OTG is flaky.  With a USB Battery/Charger connected to the UART micro USB connecting a CGM receiver with an OTG adapter cable to the OTG micro USB succeeds about 50% of the time.  When it fails, the device has not been recognized by jubilinux.  A similar issue occurs with the Intel Breakout Board if only using USB power, however with Vbat applied (>5V) the issue does not occur in that setup.  Is it possible to connect an external battery to the explorer board, and what is the allowable voltage range and pins (D?) to use.  Has OTG testing been done with just USB power applied through the UART connection?

scottleibrand, 2016-12-26:
Are you asking about connecting an external battery via direct solder points on the board?  We've supplied 5V power via the microUSB UART port, and confirmed that OTG can be used, but we've also sen flakiness.  I've seen the Dexcom charging icon light up and `lsusb` still fail, which seemed to me to indicate OTG flakiness rather than power, but I suppose a direct 5V connection would be a good way to test that.  @mogar do you know if that's possible, and if so which holes we'd want to wire 5V to?

glenrob1940, 2016-12-27:
Yes, that is what I am asking

mogar, 2016-12-27:
There aren't any test-points on the board for 5V, but if you're handy with soldering then you'll be able to do it. You could solder a 5V connection to the cathode of D5 (right next to the OTG USB port). That would allow you to power the OTG device directly.

Another thing you could do is to use the UART USB jack to apply power, but then just cut the USB cable and supply your own 5V source.

glenrob1940, 2016-12-27:
Hi Scott,

I was asking about that, but also whether or not we could use a higher 
voltage external power source via direct solder points on the board.  
The Intel breakout board allows 7-15 Volts on it's external power input 
(which is down-converted to 5 volts).  When I run my edison/intel 
breakout board, I plug in a 9V or 12 V external power source. I also 
have a LIPO, but it of course can't power USB. A USB charger/battery can 
be plugged into the micro USB UART port, but having another wired in 
power point would allow other battery choices, and might cure the OTG 
flakiness.

Glen

mogar, 2016-12-27:
The maximum voltage on the 5V rail is 5.5V. Higher than that could damage the UART-USB converter.

If you apply 5V directly to the cathode of D5, then you'll be powering the OTG device with exactly 5V (and not have the diode drop to account for).

glenrob1940, 2016-12-27:
Hi Morgan,

I also replied to Scott.  One of the (not articulated) questions I had 
was whether it was possible to apply higher voltage.  The Intel mini 
breakout board has a 7-15V external power input which it down-converts 
to 5V.  I did not know if your board had similar external power input.  
I will investigate implementing one of your two suggestions.  Thanks,

Glen

glenrob1940, 2016-12-27:
Okay.  I will keep that in mind.

Glen

closed 2017-01-08 by mogar