EnhancedRadioDevices/915MHzEdisonExplorer #12: Reversed LiPo fries the charging circuit

sulkaharo, 2017-01-07:
Just discovered I'd been shipped JST connectors with reversed current markings: a battery was connected with reversed ground/hot for around one second, after which smoke appeared and the Edison now crashes if a batter is connected to the Explorer. USB power seems to work. Might be worth checking if a future iteration of the board can include a reverse current/battery protection circuit.

mogar, 2017-01-08:
Thanks for the report. I'll add this to the list of improvements to make for the next revision. We'll probably do another revision in the second half of this year.

bmammon, 2017-06-15:
I think this might be my problem.. @sulkaharo, when you connected the battery to the Explorer, did the red power light keep blinking? When connected to USB power, everything runs fine so I think current was reversed. Is there a way to fix this? @mogar