EnhancedRadioDevices/915MHzEdisonExplorer #13: Alpha board antenna orientation

cjo20, 2017-01-11:
I've been looking in to some comms issues, and I noticed that the orientation of the antenna on my alpha board doesn't seem to match the datasheet / production boards (assuming this is the correct data sheet: http://www.johansontechnology.com/datasheets/antennas/0915AT43A0026.pdf)

The brown mark is towards the outside edge of my board. 

I am managing to get comms working, but it seems to have intermittent issues with timeouts or bad CRCs. Is it possible the antenna orientation is contributing to the problem?

mogar, 2017-01-13:
You're correct that the antenna is installed backwards on the alpha boards. My testing shows that this doesn't have much of an effect. I doubt this is causing your issue. If you have a soldering iron, it shouldn't be too hard to swap the antenna.

One way you could test this is to place the Explorer and your other device right next to each other. If they still have problems, then it's most likely due to a protocol issue instead of signal strength.

Is this related to your testing of the SW issue? https://github.com/EnhancedRadioDevices/915MHzEdisonExplorer_SW/issues/9

cjo20, 2017-01-13:
The thing that prompted this question is READ_PACKET timing out for no apparent reason. Increasing the timeout to 1 second doesn't improve the issue, so I'm still working on finding it. I do still get the issue when they're right next to each other. It's a bit random as to when it happens, so I'm struggling to track it down. 

I seem to have pretty much completely eliminated the other issue with some software changes. I'm waiting for comments / review from others before saying that for sure that the issue can be managed in mmeowlink.

closed 2017-01-24 by mogar