EnhancedRadioDevices/915MHzEdisonExplorer #15: serial ports on cc111x

dclovesdc1, 2017-02-06:
working with edison, explorer and pingrf sw. My understanding is that when connecting to edison, spi would be the primary communication path. Are the serial ports on cc111x wired to communicate with edison in addition? Trying to see what would be a good way to print out debug prints from firmware. Thanks.

mogar, 2017-02-07:
If you're using the subg_rfspy firmware as a basis, you can use the spi_serial library for debug print statements. You'll just have to poll the spi bus using spi_serial on the Edison.

The SPI interface on the CC1110 is USART1, which can also be used as a UART if you want. UART1/UART uses P1.6 and P1.7 as TX and RX, respectively. If you want to use UART0 instead, USART0/UART uses P1.4 and P1.5 as TX and RX, respectively.

dclovesdc1, 2017-02-07:
Thanks for the info. I am using pingrf. I am trying to see if I could get the spi running on USART1 (alt2) while running UART to print out debugging information. So, it seems like I could use UART0, which is also alt2 configuration.
It looks like another option would be to run UART0 at alt2 & UART1 at alt2.
Can you please let me know if I am interpreting what you are saying correct?
Thanks.

dclovesdc1, 2017-02-07:
one more thing I forgot to ask, if I use usart0/uart & usart1/uart configuration, which device would that be associated on the edison? Thanks.

mogar, 2017-02-08:
On the Edison, the pins those are connected to are set up as SPI. You could implement a software serial library to handle them though. For Edison pin information, check out the mraa guide here: https://iotdk.intel.com/docs/master/mraa/edison.html

You could also try getting spi_serial working with pingrf. Check out the spi1_alt2 code in subg_rfspy to see how that works. Then you could use the pingrf firmware with our spi_serial Edison library.
https://github.com/EnhancedRadioDevices/subg_rfspy/tree/master/spi1_alt2

dclovesdc1, 2017-02-08:
Thanks for the info. I haven't worked with mraa before. So, I'll need to look into it.
working with spi1_alt2 mode for the general communication is fine. but i am also interested in using the serial port concurrently to print out debug information from the firmware. That is actually how pingrf works today (usb for communication & uart1 for debug, usart0 for communication & uart1 for debug, etc) Any examples that you can suggest to enable uart0 for debug print while using spi1_alt2 example for the general communication? Thanks.

mogar, 2017-02-10:
Unfortunately, the serial port and the SPI port use the same pins. You'll have to choose one or the other. We can look into opening up the other UART on the next hardware revision.

dclovesdc1, 2017-02-11:
Ok. Any examples or suggestion on how to choose the serial port instead of spi? Thanks.

dclovesdc1, 2017-02-14:
Thanks for the info. Do you know which /dev/?? will be connected to the serial port if I enable serial port only on cc1110? Thanks.

mogar, 2017-02-20:
The CC1110 comms pins are connected to a SPI port on the Edison. You can interface with those pins as GPIO, but I don't think they have a hardware serial port attached to them. You'd need to write a software serial driver that manages those pins as GPIO. That shouldn't be too tough if you just set up interrupts on the RX pin. 

To see how to set the CC1110 comms pins in different modes, take a look at the subg_rfspy source:
https://github.com/EnhancedRadioDevices/subg_rfspy

Specifically, the code in the spi1_alt2 directory sets up the pins as a SPI bus. You can also build the project with uart0 or uart1 enabled. The makefile should show you how to do that.

Take a look at the schematic for pin connections between the CC1110 and the Edison:
https://github.com/EnhancedRadioDevices/915MHzEdisonExplorer/blob/master/EdisonBlock_915MHz_fab/EdisonBlock_915MHz.pdf