EnhancedRadioDevices/915MHzEdisonExplorer #16: problems with USB OTG trace cut

PieterGit, 2017-02-19:
I cut the trace as described on the wiki
https://github.com/EnhancedRadioDevices/915MHzEdisonExplorer/wiki#usb-otg-flakiness
but I still don't get USB working on a Dexcom G4 receiver.

```
# lsusb
unable to initialize libusb: -99
# dmesg | tail
[   80.954248] pmic_ccsm pmic_ccsm: USB ID Detected. Notifying OTG driver
[   81.923003] dwc3-device dwc3-device.1: device suspended; notify OTG
# uname -a
Linux lou77 3.10.17-poky-edison+ #6 SMP PREEMPT Wed Mar 23 21:47:59 EDT 2016 i686 GNU/Linux
```

Does `lsusb` work with other boards or was my cut not deep enough?
Any other things I could try?

PieterGit, 2017-02-19:
Distribution: jubilinux, updated today 
Powered with 5V on UART port.

PieterGit, 2017-02-19:
tried it on same explorerboard with other edison with more recent kernel. that works!

# uname -a
Linux myedison 4.8.0-rc4 #3 SMP Mon Sep 5 07:49:17 EDT 2016 i686 i686 i686 GNU/Linux
# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 22a3:0047
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Unfortuately communicating with the CC1111 on a newer kernel does not seem to work. Any way to get it working with a linux 3.10 kernel?

mogar, 2017-02-20:
Cutting the trace is a bit of a hack, and doesn't completely fix the problem. I know it works with the recommended Jubilinux image, but I can't guarantee anything else. The next set of Explorer boards have this issue actually fixed.

This may also be an Edison issue. Are you sure that the kernel variant you're using can access the Edison's USB port under normal circumstances? Do you have a different dev-board you can use to test the Edisons?

How come you're not using the standard Jubilinux/MRAA/spi_serial method?

PieterGit, 2017-02-20:
It's the other way around. USB OTG doesn't work with the recommended Jubilinux image for me with an Explorer board with trace cut. But it does work with a 4.8 ubuntu image. That's why I tried to upgrade the kernel, to see if that fixes the USB OTG problem. It did (after the trace cut), but unfortunately the  4.x kernel does not work nicely with the CC1110 https://github.com/EnhancedRadioDevices/915MHzEdisonExplorer/issues/17

mogar, 2017-02-22:
Gotcha. In that case I think your trace cut may not have been complete. I can't say with certainty without seeing your board, but I think that some of your issues may be due to incorrectly modified hardware. The best thing to do at this point is probably to email me directly at sales@enhancedradio.com.

PieterGit, 2017-04-09:
Tracecut was succesfull. The 3.10.17 kernel is not very suitable for using USB subsystem. Better to upgrade the kernel 3.10.98. I can get the Dexcom G4 with USB working with jubilinux 0.1.1 when I append
```
rmmod g_multi
modprobe g_multi
```
to `/etc/rc.local`. 

Unfortunately the USB OTG only works after replugging the USB OTG cable on the Explorer board side (not on the dexcom side). Then a `lsusb` will show
```
# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 22a3:0047
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

@mogar or @scottleibrand : 
Can the Explorer board detect that the OTG cable is released? I don't understand why pulling the plug at the Dexcom reciever side doesn't work. Is this a hardware issue that has been fixed with the newer Explorerboard revisions described in https://github.com/EnhancedRadioDevices/915MHzEdisonExplorer/issues/14 ?

mogar, 2017-05-05:
Latest hardware revision fixes this issue. Please contact us directly if you continue to have issues with it.

closed 2017-05-05 by mogar