EnhancedRadioDevices/915MHzEdisonExplorer #18: Change R11 value from 330 to 1.21k

jseaber, 2017-05-12:
R11 drives the Edison power LED, D6, consuming about 4mA, assuming a typical Vf of 2.1V for a red LED.

The LED is quite bright, so operating at lower current will improve battery run time (helpful for OpenAPS)! Most LEDs are still sufficiently bright at 1mA.

The same change could be applied to all LED resistors to maintain BOM simplicity (R1, R11, R9, R10).

scottleibrand, 2017-05-12:
I'm not sure how significant the battery life impact would be in real-world use, but regardless of that I think most of us running OpenAPS rigs at the bedside would prefer dimmer LEDs so we can more easily avoid light-in-the-face.  :-)

jdslabs, 2017-05-12:
Agreed! The LED currently lights up my wall at night. 

Assuming 16 hours run time from a 2000mAH battery, saving 3mA*16 hrs = 48 mAh. It's a small amount, but should boost run time by 20min per charge.

mogar, 2017-05-15:
No problem. I'll talk to the manufacturer about this. Due to lead time on boards, it may take us up to two months to have boards with the higher resistance value.