EnhancedRadioDevices/915MHzEdisonExplorer #20: Suggestion re red LED on Explorer board - the "in operation" LED, NOT the charging LED

mhaeberli, 2017-05-20:
Thanks for the Explorers!

If possible, I'd like to suggest for future Explorer boards that the red LED that shows the Explorer/Edison is on and functioning be changed to an RGB LED so that the Edison firmware could show additional status using LED color.  For example, in an openaps application, green could mean "all is working, loop is running"...

mogar, 2017-05-20:
The red LED is only connected to power, and is on whenever there is power to the board.

There are also two User LEDs that can be controlled from the Edison programatically. One is red, and one with green. You can hook those into whatever functionality you want in software, thought that will lead to a slight decrease in battery life to have them on rather than off.

mhaeberli, 2017-05-20:
Hi!
Where can I find hints on the two user LEDs?
Suggest / request RGB LEDs for future?
Best,
Martin

scottleibrand, 2017-05-20:
LED control is done in the subg_rfspy firmware.

mhaeberli, 2017-05-20:
Scott - thx, but is there any simple API to fiddle LEDs from the openaps code, like for example some kind of python bindings?  Or would I have to build custom subg_rfspy firmware to this end (referring to the red and green LEDs mentioned)?  I don't want to upset the rest of the delicate balance...

scottleibrand, 2017-05-20:
Nope. Someone would have to build that into subg_rfspy.

mhaeberli, 2017-05-20:
ok, thx

mogar, 2017-05-20:
It is built into subg_rfspy. There is a simple command to turn LEDs on or
off. Check out the examples in that repo.

closed 2018-03-08 by mogar