EnhancedRadioDevices/915MHzEdisonExplorer #23: Blinking power light/ Damaged JST?

rainbow-lazer, 2018-06-15:
Hello, I recently got a new lipo battery for my board, the JST connector of which got stuck in the port. I had to pull it out (as gently as possible) with pliers. 

After soldering on a new JST, my board's power light (D6) and one of the radio lights (D2), blink on and off, and the board doesn't seem to boot off the battery anymore. Here's a video: [https://www.dropbox.com/s/ttkknapebdbdmxr/2018-06-15%2014.51.54.mov?dl=0](https://www.dropbox.com/s/ttkknapebdbdmxr/2018-06-15%2014.51.54.mov?dl=0)

When I connect via USB, the lights work as usual and it boots fine. The receiving power port on the board looks just fine, I don't see any bent pins or cracked solder; is there any other way to troubleshoot what's going on with my JST power? 

Thanks!

rainbow-lazer, 2018-06-15:
ps, just to troubleshoot, I used a multimeter on the repaired battery and it is giving 3.7v. I also tried the battery from my backup board (which I know works), and still get the same blinking lights.

mogar, 2018-07-03:
Can you email us at sales@enhancedradio.com? We can work on this issue there.